KPI-Rover/ecu_sw #12: Призначити виводи мікроконтролера

AksonovSergei, 2024-10-20:
Для початку розробки друкованої плати необхідно призначити виводи мікроконтролера та розробити мінімальну програму, яка дозволить перевірити працездатність на відладній платі. Після цього ця програма буде використовуватись для перевірки друкованої плати.
Послідовність виконання завдання:

1.  Ознайомтеся з вимогами до друкованої плати, щоб визначити сигнали, якими має керувати мікроконтролер.
2.  Створіть проєкт у STM32Cube IDE і налаштуйте всі необхідні входи та виходи.
3.  Перевірте працездатність кожного сигналу на відладній платі.
4.  Оформіть у цій задачі таблицю мапінгу сигналів до виводів мікроконтролера.
5.  Створіть Pull Request.

AksonovSergei, 2024-11-10:
@vetshem69 
Я додав Ваш .ioc file до бранчу dev.
В цілому все добре але Ви планували використовувати вивода SWD інтерфейсу (PA13, PA14) - це не дуже добре залишитись без можливості покрокової відладки плати. Я переніс Buzzer на PD10.
[image]

Будь ласка зробіть повні назви для виводів (User Label). Наприклад як я зробив для PB5, PB4.
Правки робіть будь ласка у бранч dev

AksonovSergei, 2025-03-02:
Hi @Rhinemann and @nazarbychkar,
How's it going? If you have any questions or need support, please ask in this task.

Rhinemann, 2025-03-06:
# MCU Pin labels and types

## Motor control (4 groups)
- Motor_PWM_(1...4) - timer-generated PWM
- Motor_Fwd_(1...4) - GPIO_Output
- Motor_Rvr_(1...4) - GPIO_Output
- Motor_Gnd_(1...4) - not on the MCU

## Encoder outputs (4 groups)
- Encoder_Pow_(1...4) - GPIO_Output or just non-MCU power?
- Encoder_Gnd_(1...4) - not on the MCU
- Encoder_A_(1...4) - GPIO_Input?
- Encoder_B_(1...4) - GPIO_Input?

## Blue stasus LEDs
- Blue_LED_(1...4) - GPIO_Output

## External status LEDs
- Red_LED - GPIO_Output
- Yellow_LED - GPIO_Output
- Green_LED - GPIO_Output

## Serial interface connector (USART?)
- Serial_Connector_TX - USART[123]_TX
- Serial_Connector_RX - USART[123]_RX

## Ultrasound sensor inputs (2 groups)
- US_Sensor_Pow_(1...2) - GPIO_Output or just non-MCU power?
- US_Sensor_Gnd_(1...2) - not on the MCU
- US_Sensor_Trig_(1...2) - GPIO_Output?
- US_Sensor_Echo_(1...2) - GPIO_Input?

## Serial interface connector for GPS (USART?)
- GPS_Connector_TX - USART[123]_TX
- GPS_Connector_RX - USART[123]_RX

##  I2C interface connector for external compass
- TBD

AksonovSergei, 2025-03-07:
Hi @Rhinemann,
Thank you for your effort. Here are the answers to your questions. Let me know if you have any further questions.

>Encoder_Pow_(1...4) - GPIO_Output or just non-MCU power?

It must be connected `+3.3V` power source and should not be connected to GPIO.

> US_Sensor_Pow_(1...2) - GPIO_Output or just non-MCU power

It must be connected `+5V` power source and should not be connected to GPIO.

> Encoder_A_(1...4) - GPIO_Input?
> Encoder_B_(1...4) - GPIO_Input?

Its must be GPIO inputs connected to a TIMER that has `Encoder Mode.`
STM32 MCUs have dedicated hardware support for reading quadrature encoders using `Timer Encoder Mode`. This mode allows you to read encoders efficiently without polling GPIOs manually.

>US_Sensor_Trig_(1...2) - GPIO_Output?

It should be assigned to GPIO in the `Output Push-Pull mode`.

>US_Sensor_Echo_(1...2) - GPIO_Input?

This signal must be connected to a GPIO that supports `TIMx in Input Capture Mode`.
**How It Works**

1. MCU sends a 10µs pulse on the Trig pin.
2. Sensor sends back an Echo pulse.
3. MCU uses Timer Input Capture Mode to measure pulse width.
4. Convert time to distance using Speed of Sound formula.

Rhinemann, 2025-03-17:
Great, thanks for the answer, that means we have this pin arrangement:
# MCU Pin labels and types

## Motor control (4 groups)
- Motor_PWM_(1...4) - timer-generated PWM
- Motor_Fwd_(1...4) - GPIO_Output
- Motor_Rvr_(1...4) - GPIO_Output
- Motor_Gnd_(1...4) - not on the MCU

## Encoder outputs (4 groups)
- Encoder_Gnd_(1...4) - not on the MCU
- Encoder_A_(1...4) - Timer Encoder Mode GPIO
- Encoder_B_(1...4) - Timer Encoder Mode GPIO

## Blue stasus LEDs
- Blue_LED_(1...4) - GPIO_Output

## External status LEDs
- Red_LED - GPIO_Output
- Yellow_LED - GPIO_Output
- Green_LED - GPIO_Output

## Serial interface connector (USART?)
- Serial_Connector_TX - USART[123]_TX
- Serial_Connector_RX - USART[123]_RX

## Ultrasound sensor inputs (2 groups)
- US_Sensor_Gnd_(1...2) - not on the MCU
- US_Sensor_Trig_(1...2) - Output Push-Pull mode
- US_Sensor_Echo_(1...2) - TIMx in Input Capture Mode
## Serial interface connector for GPS (USART?)
- GPS_Connector_TX - USART[123]_TX
- GPS_Connector_RX - USART[123]_RX

##  I2C interface connector for external compass
- TBD

Also, I have a question, what about the Ethernet connection, as it isn't specified in the documentation?

AksonovSergei, 2025-03-18:
Hi,
We need Ethernet interface if it is possible. I have added requirement.